WorldFamousElectronics/PulseSensorStarterProject #2: Can I run on Arduino 101?

frankthedead, 2017-04-28:
...

biomurph, 2017-04-28:
Hi frankthedead,

The short answer is YES!
The PulseSensorStarterProject will work just fine on the Arduino 101.

If you want to use the PulseSensorAmped 1.5.0 code, you will have to modify in order to NOT use the hardware timer. Instead, use a software timer, for which example code is shown in the Time_Interrupt_Notes.ino tab in the code.

andreaRo1997, 2017-09-05:
Hi biomurph
I´am from Ecuador and  I and a group of university students are using the pulse sensor with this code.
 We connect a buzzer to the Arduino, It runs properly, but it sounds also when doesn´t sense any pulse.
In case of this proyect  we want that the sensor only sounds when it detects pulse, is possible do that? we´re trying to use the pulse sensor as a pulse detector. Thanks for the attention, 
greetings from Ecuador.

biomurph, 2017-09-05:
@andreaRo1997 
I'm responding in regard to this PulseSensorStarterProject code.
The reason that you are having this problem is because of noise. You can avoid this noise by increasing the value of the variable called threshold. You should experiment with it. If you raise it too high, you might miss the pulse. If it's too low, you will get more noise.

Pipo3060, 2018-06-20:
Hello, 
I was wondering if there are any projects on making this library compatible with teensy boards or some WiFi micros like the huzzah from adafruit. I tried the different examples for the BPM  and even the code that doesn't use interrupts and I can get them to compile for any of these boards. If you are not planning on adding the support for these boards can you please point me what can I do to make it work ? 
thanks !

biomurph, 2018-06-21:
@Pipo3060 This issue is more appropriately posted in the PulseSensor Playground repository.
The BPM_Alternative sketch found in the Playground will compile for modules using ESP8266.
It will also compile for the Teensy family of boards. Please go here
https://github.com/WorldFamousElectronics/PulseSensorPlayground
Or use the Libraries Manager window in Arduino to install the latest version.

closed 2018-06-21 by biomurph