WorldFamousElectronics/PulseSensorStarterProject #4: I somehow can't get any good data

victoroos, 2017-10-18:
Hi

After connecting the heartbeat sensor, I'm now trying to get data from it. However, it just spikes a bit, not even looking like a beating heart. The sensor lights up green so it is connected..
Any tips?
vkr
victor

biomurph, 2017-10-18:
Victor,
I will need more information from you to help troubleshoot this problem. Can you send a picture of the sensor? What Arduino board are you using? How are you visualizing the data?

victoroos, 2017-10-18:
sensor as the hardbeat sensor? SUre, will do in a few. For the other questions: Uno. And using the serial plotter as instructed.

biomurph, 2017-10-18:
also send a screen shot of the data in the serial plotter

victoroos, 2017-10-19:
[image]

victoroos, 2017-10-19:
[image]

victoroos, 2017-10-19:
Can you break the sensor when attach it wrong on accident, because that is what the store where I bought it is telling me. I was confused by the grey and white, which both were neutral I thought.

biomurph, 2017-10-19:
Well, it looks like you purchased a counterfeit Pulse Sensor. That is not our PCB, nor is it our wire type and color. I'm sorry, but since I don't have any idea what's going on with that hardware I can't troubleshoot this problem for you. 

Please support the Original Makers of Pulse Sensor by purchasing from a legitimate reseller.
www.pulsesensor.com
www.adafruit.com
www.sparkfun.com

victoroos, 2017-10-19:
I didn't even know that was an option, you might wanna contact https://www.hackerstore.nl/ about it, they are a legitamate company, maybe they don't know about it either, but yes I will support you guys
Thanks! :)

closed 2018-01-10 by yury-g