WorldFamousElectronics/PulseSensorStarterProject #5: I have a WEMOS D1R1 that runs on 12 Mhz clock oscillator

Drocerika, 2017-12-11:
I did change the code as showed in Time_Interrupt_Notes.ino but the pulse rate is reading too fastly.
What can I do to make it slower or can I use some kind of hardware Timer like in Arduino?
The code runs perfectly on my Arduino mega but I need to run it on wemos D1R1 because it has ESP8266 module implemented and after that I want to get the data on a webpage.

yury-g, 2017-12-11:
Can you send code snippet change you made ?  That may help....

Drocerika, 2017-12-11:
Here it is.

[PulseSensorAmped_Arduino_1.5.0.zip](https://github.com/WorldFamousElectronics/PulseSensorStarterProject/files/1548723/PulseSensorAmped_Arduino_1.5.0.zip)

yury-g, 2018-01-10:
Drocerika, 
We just published a Library for PulseSensor and Arduino.  Take a look at the code:
[https://github.com/WorldFamousElectronics/PulseSensorPlayground](https://github.com/WorldFamousElectronics/PulseSensorPlayground
)

Trying giving this an installation. It has a lot of functions in easy reach. 

The Function Guide for the PulseSensor Arduino Library is here:
https://github.com/WorldFamousElectronics/PulseSensorPlayground/blob/master/resources/PulseSenaor%20Playground%20Tools.md

closed 2023-10-25 by erikaess