WorldFamousElectronics/PulseSensorStarterProject #9: Pulse Sensor giving unexpected readings based on examples

RoboUser, 2018-02-25:
Hello,

I appear to be coming across an issue using the pulse sensor shown in this repository and the library doesn't seem to clear things up for me. I have tried reading my pulse sensor using the visualiser but get very noisy readings with very low amplitude.

The code I have employed is the following, but I have also tried running through the PulsePlayground library and am obtaining the same results:
`#include <Adafruit_GFX.h>
#include <UTFTGLUE.h>
UTFTGLUE myGLCD(0x9488, A2, A1, A3, A4, A0);

// Declare which fonts we will be using
#if !defined(SmallFont)
extern uint8_t SmallFont[];    //.kbv GLUE defines as GFXFont ref
#endif

#define UpperThreshold 516
#define LowerThreshold 515

void setup() {
  Serial.begin(9600);
  pinMode(A0, OUTPUT);       //.kbv mcufriend have RD on A0
  digitalWrite(A0, HIGH);
  // Setup the LCD
  myGLCD.InitLCD();
  myGLCD.setFont(SmallFont);
  // Clear the screen and draw the frame
  myGLCD.clrScr();
  // BPM Parameters
  myGLCD.setColor(255, 255, 255);
  myGLCD.print(" BPM", 5, 34);
}

void loop() {
  
int LastTime = 0;
bool BPMTiming = false;
bool BeatComplete = false;
int BPM = 0;

  static unsigned long lastTime = 0;
  const unsigned long INTERVAL = 20; //milliseconds, calculated as 1 divided by sample frequency
  while (millis() - lastTime < INTERVAL)
  {
    //wait; do nothing until the sample interval expires
  }
  lastTime = millis();

  // Calculate when a beat has been completed
  if (analogRead(A5) > UpperThreshold)
  {
    if (BeatComplete == true)
    {
      BPM = millis() - LastTime;
      BPM = int(60 / (float(BPM) / 1000));
      BPMTiming = false;
      BeatComplete = false;
    }
    if (BPMTiming == false)
    {
      LastTime = millis();
      BPMTiming = true;
    }
  }

  if ((analogRead(A5) < LowerThreshold) & (BPMTiming))
  {
    BeatComplete = true;
  }

  Serial.println(analogRead(A5));   //Plots the readings in the serial plotter

  // Display BPM when calculated at the bottom of the loop
  myGLCD.setColor(255, 255, 255);
  myGLCD.printNumI(BPM, 40, 34);
}`

I have attached an image of the readings I'm getting and am hoping you might be able to give me an insight into the reason for the difference. The image on the left is from: https://pulsesensor.com/pages/pulse-sensor-amped-arduino-v1dot1

[image]

biomurph, 2018-02-26:
I do not really understand your code. Not sure how 
while (millis() - lastTime < INTERVAL)
is ever true when lastTime is always 0 ??
You say you ran the Playground example, and you get the same noisy signal.
First, make sure that you unplug your computer from the mains power. That can sometimes cause noise in the signal. Your signal is very small,however, so that makes me think that is not the only problem. Some questions.

First, what OS are you using? What microcontroller board are you using?
Second, please take a picture of the Pulse Sensor, so that I can verify that it is our hardware.

RoboUser, 2018-02-28:
I ought to have expanded more. That particular piece of code was suggested by another user and I included it to see if it would help me.
[image]
[image]

 The following block of code has no effect on the readings:

```
static unsigned long lastTime = 0;
const unsigned long INTERVAL = 20; //milliseconds, calculated as 1 divided by sample frequency
while (millis() - lastTime < INTERVAL)
{
//wait; do nothing until the sample interval expires
}
lastTime = millis();
```

- Interesting that you mentioned that the mains power could cause noise. I have included an image of the reading with the power supply unplugged. It's using the same code, but looks very different. I tried holding the sensor various ways but it never got any better. I tried plugging in the mains again and the waveform changed to the same one you see in my first post.

- My OS is Windows 10 Home 64 bit

- The microcontroller I am using is the Arduino Uno R3

- I have included a picture of the Pulse Sensor.

biomurph, 2018-02-28:
Well, there's your problem right there. You are using a counterfeit Pulse Sensor. That is not our hardware, and I cannot even begin to troubleshoot since I have no idea what that thing is made of.

Please support the original makers of Pulse Sensor by purchasing the Original Pulse Sensor.
www.pulsesensor.com

closed 2018-02-28 by biomurph

RoboUser, 2018-02-28:
I had no idea. I thought I was purchasing a legitimate one. I will order a proper one immediately and hopefully that will sort out my problem. Thank you for letting me know.

biomurph, 2018-02-28:
where did you get it?