WorldFamousElectronics/PulseSensorStarterProject #11: Library not working for Arduino Uno WiFi Rev2.

Tommaarleveld, 2019-03-18:
It seems the code can't be compiled for the Arduino Uno WiFi Rev2. Is there a simple fix for this issue? Or do I need to use a regular Arduino Uno?

This is the error I got:
```
Arduino: 1.8.9 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

In file included from /Users/tommaarleveld/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.25/cores/arduino/Arduino.h:27:0,
                 from sketch/Getting_BPM_to_Monitor.ino.cpp:1:
/Users/tommaarleveld/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h: In function 'void TIMER1_COMPA_vect()':
/Users/tommaarleveld/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/Interrupts.h:207:9: warning: 'TIMER1_COMPA_vect' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]
     ISR(TIMER1_COMPA_vect)
         ^
/Users/tommaarleveld/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp: In member function 'void PulseSensor::updateLEDs()':
/Users/tommaarleveld/Documents/Arduino/libraries/PulseSensor_Playground/src/utility/PulseSensor.cpp:221:33: error: cannot convert 'volatile boolean {aka volatile bool}' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'
     digitalWrite(BlinkPin, Pulse);
                                 ^
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

```

biomurph, 2019-03-18:
@Tommaarleveld 
I just published a new release of the PulseSensor Playground library.
Please download it here
https://github.com/WorldFamousElectronics/PulseSensorPlayground/releases
or use the Arduino Library Manager to update. This should fix your problem. LMK

Tommaarleveld, 2019-03-18:
@biomurph Thanks for the quick reply! I updates the library and it seems my first error has been fixed. But when I try to upload the example Getting_BPM_To_Monitor and test the sensor I get the following message in the serial monitor: "interrupts not supported".

In [this](https://github.com/WorldFamousElectronics/PulseSensorPlayground/issues/45) thread I found a similar problem. So I tried out the code snippet which you suggested.

> We don't yet support use of interrupts on the Mega. You will need to run the BPM_Alternative code example. If all you want is BPM on the serial port, then you will need to add a variable at the top of the sketch
> 
> int myBPM;
> 
> Then modify part of the code in the loop to be something like this
> 
> if (pulseSensor.sawStartOfBeat()) { myBPM = pulsesensor.getBeatsPerMinute(); Serial.print("BPM: "); Serial.println(myBPM); }

Now the serial monitor starts to output unreadable characters:
[image]

Sidenote: While uploading or verifying the code, it does output a few warnings but compiles anyways.

Hope you can help me with this!

Tommaarleveld, 2019-03-18:
So apparently this was due to the wrong baud setting. I changed it to  115200 baud and now it sort of works. Although I'm getting an insane amount of readings with occasionally the one that I'm asking for:
[image]

biomurph, 2019-03-18:
@Tommaarleveld 
ahh, baud rate mismatch... a rookie mistake ;)

Yes, it does look like you are getting the right data. The way the sketch is set up is to send the data as 
BPM,IBI,Signal
once every 20mS
If you want to ignore the data stream, comment out the line that says `pulseSensor.outputSample()`
Then, in the conditional `if(pulseSensor.sawStartOfBeat())` just put in your serial statement to print the BPM.

The serial reply that says `Interrupts not supported` is there to tell you to use the BPM_Alternative code. 

We're working on good examples that use the built in library tools to monitor jitter when you aren't using interrupts. Stand by for that.

Tommaarleveld, 2019-03-19:
@biomurph 
Thanks for the help, it works pretty well now. I'll definitely stand by for the new updates!

I attached the sensor to the velcro strip. But when I strap it all around my finger, most of the time it starts out with a lot of high readings. When I hold my finger absolutely still it starts to settle down and gives the correct reading. Have you got any pointers on how to get the more  accurate readings?

biomurph, 2019-03-19:
There is kind of a 'goldilocks' pressure that you want on the sensor.
Pulse Sensor, like any PPG based heart monitor, measures the change in tissue density. When a pulse wave courses through your body at every heartbeat, your capillary tissues get less dense for just a moment. That's what is making the reflected light change in intensity. 

The thing that will also make your tissue density change is movement. We are all basically bags of blood and bone, sloshing around will make the signal change.

Also, I never asked you. Did you get a legit Pulse Sensor? Does it have black, red, and purple wires about 24" long? There are lots of cheap knockoffs out there.

Tommaarleveld, 2019-03-19:
Alright seems clear!

I'm pretty sure I do have a legit Pulse Sensor. It does have the black, red and purple wires. Although the wires are more like 15" long. It came with the velcro and steel clip. Here's a picture:
[image]

biomurph, 2019-03-19:
Yup, looks good! Thanks for supporting the original makers of Pulse Sensor!
Have fun with your project!

closed 2019-03-19 by biomurph