WorldFamousElectronics/PulseSensorStarterProject #13: RF24 Library with Arduino Uno WiFi R2

clearlincognito, 2020-01-26:
I am getting this error every time I try to compile it. Is there any way to fix this? Thank you.
`Arduino: 1.8.10 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

/private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware -hardware /Users/clearlincognito/Library/Arduino15/packages -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/clearlincognito/Library/Arduino15/packages -built-in-libraries /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/libraries -libraries /Users/clearlincognito/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=0X03EB_0X2145 -ide-version=10810 -build-path /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216 -warnings=none -build-cache /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_cache_692692 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/clearlincognito/Documents/Arduino/sketch_jan04a/sketch_jan04a.ino
/private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware -hardware /Users/clearlincognito/Library/Arduino15/packages -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/clearlincognito/Library/Arduino15/packages -built-in-libraries /private/var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/AppTranslocation/5A7D5ABA-E93B-4917-ABB2-B9AAD19288BA/d/Arduino.app/Contents/Java/libraries -libraries /Users/clearlincognito/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -vid-pid=0X03EB_0X2145 -ide-version=10810 -build-path /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216 -warnings=none -build-cache /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_cache_692692 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /Users/clearlincognito/Documents/Arduino/sketch_jan04a/sketch_jan04a.ino
Using board 'uno2018' from platform in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Using core 'arduino' from platform in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5
Detecting libraries used...
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI/src /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
Alternatives for nRF24L01.h: [RF24@1.3.3]
ResolveLibrary(nRF24L01.h)
  -> candidates: [RF24@1.3.3]
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI/src -I/Users/clearlincognito/Documents/Arduino/libraries/RF24 /var/folders/nj/1yj7skjs4qscz69d8sg3zc_h0000gp/T/arduino_build_6216/sketch/sketch_jan04a.ino.cpp -o /dev/null
/Users/clearlincognito/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino/api/deprecated -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/cores/arduino -I/Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/variants/uno2018 -I/Users/clea [... cut ...]
 Used: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI
Multiple libraries were found for "nRF24L01.h"
 Used: /Users/clearlincognito/Documents/Arduino/libraries/RF24
Using library SPI at version 1.0 in folder: /Users/clearlincognito/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/libraries/SPI 
Using library RF24 at version 1.3.3 in folder: /Users/clearlincognito/Documents/Arduino/libraries/RF24 
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.`

closed 2020-01-26 by clearlincognito

biomurph, 2020-01-26:
@clearlincognito 
Those errors are from the RF24 library. Did you get it fixed?

clearlincognito, 2020-01-27:
Unfortunately not. I accidentally posted it here instead so I posted it on the RF24 library.

biomurph, 2020-01-27:
Drat!
Well, if you have questions or issues with your Pulse Sensor, please come back.

clearlincognito, 2020-01-27:
I will :)