WorldFamousElectronics/PulseSensorStarterProject #16: PulseSensor-RaspberryPi  - BPM 3-times too high?

webster1501, 2021-04-23:
Dear team,

first of all my appreciation for what you are doing here with the PulseSensor and the code you are providing!

My setup includes a RaspberryPi 3b and I have the MCP3008 ADC sitting in between the sensor and my SPI port (8).
Everything is working fine, beside the fact, that my BPM appears approximately 3 times higher than in reality (I measured it with a commercial product).
I am providing a sample output below.

Any idea what the root cause of this might can be ?

Best
Thomas

sampleCounter | Signal | BPM | IBI | sampleIntervalMs
-- | -- | -- | -- | --
746 | 828 | 164 | 364 | 42
787 | 899 | 164 | 364 | 41
827 | 1023 | 164 | 364 | 40
866 | 1023 | 164 | 364 | 39
904 | 655 | 164 | 364 | 38
943 | 1023 | 164 | 364 | 39
981 | 600 | 164 | 364 | 38
1019 | 561 | 164 | 364 | 38
1057 | 862 | 192 | 311 | 38
1095 | 643 | 192 | 311 | 38
1133 | 1023 | 192 | 311 | 38
1171 | 927 | 192 | 311 | 38
1208 | 711 | 192 | 311 | 37
1245 | 428 | 192 | 311 | 37
1282 | 335 | 192 | 311 | 37
1318 | 752 | 192 | 311 | 36
1354 | 912 | 202 | 297 | 36
1389 | 1023 | 202 | 297 | 35
1423 | 1023 | 202 | 297 | 34
1456 | 1023 | 202 | 297 | 33
1489 | 920 | 202 | 297 | 33
1522 | 398 | 202 | 297 | 33
1554 | 880 | 202 | 297 | 32
1586 | 1023 | 202 | 297 | 32
1617 | 865 | 228 | 263 | 31
1648 | 945 | 228 | 263 | 31
1678 | 796 | 228 | 263 | 30
1708 | 335 | 228 | 263 | 30
1738 | 638 | 228 | 263 | 30
1767 | 611 | 228 | 263 | 29
1796 | 624 | 228 | 263 | 29
1824 | 368 | 228 | 263 | 28
1852 | 816 | 228 | 263 | 28
1880 | 711 | 228 | 263 | 28
1908 | 1023 | 206 | 291 | 28
1936 | 812 | 206 | 291 | 28
1963 | 1023 | 206 | 291 | 27
1990 | 1023 | 206 | 291 | 27
2017 | 1023 | 206 | 291 | 27
2043 | 1023 | 206 | 291 | 26
2069 | 632 | 206 | 291 | 26
2095 | 371 | 206 | 291 | 26
2122 | 856 | 206 | 291 | 27
2150 | 1023 | 206 | 291 | 28
2178 | 608 | 206 | 291 | 28
2205 | 568 | 206 | 291 | 27
2232 | 423 | 206 | 291 | 27
2259 | 337 | 206 | 291 | 27
2285 | 335 | 206 | 291 | 26
2311 | 625 | 206 | 291 | 26
2337 | 415 | 206 | 291 | 26
2362 | 339 | 206 | 291 | 25
2387 | 551 | 206 | 291 | 25
2412 | 929 | 119 | 504 | 25
2437 | 719 | 119 | 504 | 25
[... 253 lines cut ...]
8591 | 636 | 202 | 296 | 27
8618 | 1023 | 202 | 296 | 27
8645 | 643 | 202 | 296 | 27
8672 | 871 | 218 | 275 | 27
8699 | 1023 | 218 | 275 | 27
8726 | 719 | 218 | 275 | 27
8753 | 632 | 218 | 275 | 27
8780 | 798 | 218 | 275 | 27
8807 | 615 | 218 | 275 | 27
8834 | 671 | 218 | 275 | 27
8861 | 831 | 218 | 275 | 27
8888 | 927 | 218 | 275 | 27
8915 | 606 | 218 | 275 | 27
8942 | 895 | 222 | 270 | 27
8969 | 460 | 222 | 270 | 27
8997 | 368 | 222 | 270 | 28
9025 | 1023 | 222 | 270 | 28
9053 | 412 | 222 | 270 | 28
9081 | 1023 | 222 | 270 | 28
9109 | 675 | 222 | 270 | 28
9137 | 432 | 222 | 270 | 28
9164 | 824 | 222 | 270 | 27
9191 | 768 | 222 | 270 | 27
9218 | 467 | 222 | 270 | 27
9246 | 1023 | 197 | 304 | 28

biomurph, 2021-04-23:
@webster1501 Are you using the example code that we provide without modification? This is the Pulse Sensor + C example, correct?
It'e been a while since I've had my rig set up, but I can attempt to replicate your results.
I do believe that I made the original work using the Pi 3b. I have that and  4 here in my lab. 
We have been using the uAlarm, which at the time of making was not recommended for future designs.
It looks like the sampleIntervalMs is too long a time. 27 milliSeconds is not the target sample timing.

Also, please edit your initial post to reduce the amount of data you are listing in the table for readability of this issue. Thanks!

webster1501, 2021-04-23:
Hi,

thanks for the quick reply.
1. I reduced the the table size of my initial post.
2. I'm using the code which you provided her on GitHub:
https://github.com/WorldFamousElectronics/Raspberry_Pi/tree/master/PulseSensor_Processing_Pi/PulseSensor_Processing_Pi

I'm curious to hear if you can replicate my observation, and what redesign suggestions you may have. I have also a PI 4 in my fundus, if you think it makes sense to use this instead of my 3b?

Best
Thomas

biomurph, 2021-04-23:
Oh, you're using the Processing Sketch.
Well, that is likely not super accurate in the first place, but you may have some luck in improving by adding `P2D` to your size() funcion `size(<width>,<height>,P2D); `

In a nutshell, the accurate BPM relies on accurate sample timing. Because the Processing Sketch is running on top of the  RasPi OS, there is no telling when the sketch goes through each loop and hence, when the sample is taken from the Pulse Sensor. In this case, we're trying to fix that by using the frameRate variable that should be giving us some tighter timing of when the sample is taken. It may be better to use the `millis()` function? Don't know, haven't tested it. 

Is there anything else running on your Pi? I kind of remember that I had to close all other applications.

Also, try using the C example in the repo and see if you get better results.

webster1501, 2021-04-28:
Hi Joel,

I tried it with the C example, with the same results.
I just ordered an Arduino to see the differences between it and RasPi.
I'll post my findings here as soon as I have them.

webster1501, 2021-04-30:
Hi Joel,

me again.
I did the Pulse Sensor + Arduino + Pi setup after I received my Arduino Uno, worked like a charm immediately.

Unfortunately - for me - with the same result.
This is the output I got:
16:16:46.134 -> 230,260,387
16:16:46.134 -> 230,260,377
16:16:46.168 -> 230,260,623
16:16:46.201 -> 230,260,623
16:16:46.201 -> 230,260,623
16:16:46.234 -> 230,260,626
16:16:46.267 -> 230,260,768
16:16:46.267 -> 230,260,774
16:16:46.300 -> 230,260,775
16:16:46.300 -> 230,260,833
16:16:46.333 -> 230,260,829
16:16:46.366 -> 230,260,822
16:16:46.366 -> 230,260,816
16:16:46.407 -> 230,260,936

Don't know exactly what columns represents what value, bit I assume the 1st column (230) is the BPM.
When I watch the LED though I realize that it is blinking with an estimated frequency of 1,2 HZ which would mean a realistic BPM of around 70-90 BPM ?

Strange, isn't it ?

Best
Thomas

biomurph, 2021-04-30:
Hi Thomas,

Yes it is quite strange indeed!
AT this point, for the sake of due diligence, can you share a screenshot of the pulse wave you are getting?
Also, a pic of your hardware setup, including your Pulse Sensor.

I'm in the weeds today, but I can get setup to try and replicate your problem within a week.

webster1501, 2021-04-30:
Hi Joel,

first of all - no hurry!
For me this is just something I want to demonstrate in my company and the schedule is end of May.

Below the requested images:

[image]
[image]

Enjoy the weekend!
Best
Thomas

biomurph, 2021-04-30:
Hi Thomas,

Thanks for the pictures!
The first thing that I have to say is that you have unfortunately purchased a clone of our Pulse Sensor.
Our Pulse Sensor has Red, Black, and Purple wires, and the three holes on the edge are not plated like yours are.
Sorry to say that I won't be able to help you troubleshoot if the problem lies in the sensor functionality as I have no idea what that board is made with.

Please support the original makers of Open Source Hardware by purchasing legit parts.
You can get the Pulse Sensor on our website www.pulsesensor.com
Also on Adafruit, SparkFun, Mouser, and Digikey.

webster1501, 2021-04-30:
This makes sense to me, and I wasn't aware of that I purchased a clone of the original one. Wasn't on purpose.
I will definitely support the original makers, and will order some immediately.

Thank you so much so far for your support, I'll keep you posted once the original ones have arrived!

Best
Thomas

webster1501, 2021-04-30:
Just ordered the one from your company, and I'm a bit scared that clones are underway.

This is the one I ordered:
https://www.amazon.de/MissBirdler-Kardiometer-Heartbeat-Herzschlag-Prototyping/dp/B01MSLQQK9/ref=sr_1_1?dchild=1&keywords=missbirdler+pulse+sensor&qid=1619796609&sr=8-1

biomurph, 2021-04-30:
Clones have been among us for a long time.

We are unfortunately not represented in Amazon EU.

webster1501, 2021-04-30:
That's why I apparently didn't find the original one...

closed 2023-10-26 by biomurph