WorldFamousElectronics/PulseSensorStarterProject #17: The body of the message says all, this code is not compatible with LOLIN(WEMOS)D1 mini pro

zenith85, 2021-12-02:
c:/users/ibrah/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o: in function `_ZN21PulseSensorPlaygroundC2Ei':
C:\Users\ibrah\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.cpp:28: undefined reference to `_ZN21PulseSensorPlayground15UsingInterruptsE'
c:/users/ibrah/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ibrah\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.cpp:34: undefined reference to `_Z35PulseSensorPlaygroundSetupInterruptv'
c:/users/ibrah/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\PulseSensor_Playground\PulseSensorPlayground.cpp.o: in function `_ZN21PulseSensorPlayground5beginEv':
C:\Users\ibrah\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.cpp:56: undefined reference to `_Z35PulseSensorPlaygroundSetupInterruptv'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini Pro.

biomurph, 2021-12-03:
@zenith85 
It looks like you are using a board that is based on the ESP8266.
Please try to use the PulseSensor Playground example called BPM_Alternative.ino
Let me know if that will work.

biomurph, 2023-10-26:
Our current library works with ESP32 based boards.

closed 2023-10-26 by biomurph