WorldFamousElectronics/PulseSensorStarterProject #18: Arduino nano rp2040 connect troubleshooting

myazenko, 2023-01-13:
Hello, first thank you for your work on PulseSensor, it is a great tool. I first tried it with an arduino uno on a previous project, and everything went smoothly. However now that I've got a nano rp2040, the sketches won't compile (including the examples provided with the library). I get a bunch or errors, and looking at them I feel a bit out of my depth. Could you please help me figure it out ? 
  

In file included from D:\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:439:0,
                 from D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:16:
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp: In member function 'void PulseSensor::setThreshold(int)':
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:72:41: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                         ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:80:3: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
   DISABLE_PULSE_SENSOR_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\Documents\Arduino\libraries\PulseSensor_Playground\src/PulseSensorPlayground.h:439:0,
                 from D:\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:16:
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp: In member function 'boolean PulseSensorPlayground::sawNewSample()':
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:72:41: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                         ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:105:5: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
     DISABLE_PULSE_SENSOR_INTERRUPTS;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:73:40: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                        ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:83:3: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
   ENABLE_PULSE_SENSOR_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:73:40: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                        ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\PulseSensorPlayground.cpp:108:5: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
     ENABLE_PULSE_SENSOR_INTERRUPTS;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp: In member function 'boolean PulseSensor::sawStartOfBeat()':
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:72:41: error: 'cli' was not declared in this scope
 #define DISABLE_PULSE_SENSOR_INTERRUPTS cli()
                                         ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:108:3: note: in expansion of macro 'DISABLE_PULSE_SENSOR_INTERRUPTS'
   DISABLE_PULSE_SENSOR_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Documents\Arduino\libraries\PulseSensor_Playground\src/utility/Interrupts.h:73:40: error: 'sei' was not declared in this scope
 #define ENABLE_PULSE_SENSOR_INTERRUPTS sei()
                                        ^
D:\Documents\Arduino\libraries\PulseSensor_Playground\src\utility\PulseSensor.cpp:111:3: note: in expansion of macro 'ENABLE_PULSE_SENSOR_INTERRUPTS'
   ENABLE_PULSE_SENSOR_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

biomurph, 2023-01-13:
@myazenko We try to use hardware timer interrupts when we can. The RP2040 is not supported for hardware interrupts, and is rather new to us. That is why you are getting this error. I made changes to the PulseSensor Playground library. Please use the latest release, which you can find via the Arduino Libraries Manager, or you can download from here

https://github.com/WorldFamousElectronics/PulseSensorPlayground/releases

What you need to do is use the Example called PulseSensor_BPM_Alternative.ino
That one uses a software timer to acquire the signals.

biomurph, 2023-10-26:
Our current library version supports RP2040.

closed 2023-10-26 by biomurph