WorldFamousElectronics/PulseSensorStarterProject #20: application to stm32f103r8t6

CesarPavon21, 2024-11-03:
I need help to implement the library with the stm32f103r8t6 through arduino ide.
The bpm value gives me zero, I think the problem is in the sampling time but I don't understand how to change it.

biomurph, 2024-11-04:
@CesarPavon21 
Thanks for using PulseSensor.
I need some more information  in order to help you.
Please tell me what arduino board you are using? What version of Arduino IDE are you using?
Also, please send a picture of your hardware setup so that I can verify it.

CesarPavon21, 2024-11-06:
Hello!
Thank you very much for your response.
I’m using the STM32F103R8T6 microcontroller and Arduino IDE 2.3.3.
I managed to solve it. I used a timer that generates interrupts every 2ms, but I have another question: in this way, it doesn't detect a bpm of zero at the moment when I stop placing my finger on the sensor.
Do you have any recommendations to improve real-time measurement?

biomurph, 2024-11-06:
@CesarPavon21 
Glad you got it working.
Please share your code so that we can work your fix into our library?

I will need more information to help you. Please share your code?
Also, I am not quite sure what you mean when you say “doesn’t detect bpm of zero”.
Could you explain that more?

CesarPavon21, 2024-11-08:
Understood. Thank you very much for the advice and for the continued response.